elephantrobotics/mycobot_ros #17: Please do not suggest using `pip install --user` 😬

eacousineau, 2021-05-09:
Howdy! I'm tinkering with the robot now, and having fun!

That being said, things like `sudo pip install` and `pip install --user` can wreak havoc on an existing system, and kludge things on your `PATH` and `PYTHONPATH`. On multiple instances, I've had to fix things that were borked by shadowing, and/or help colleagues do the same.

I've tried the following with `virtualenv`, and it seems to work A-OK:
```sh
sudo apt install python3-virtualenv   # Ubuntu

cd ~/catkin_ws

# Do nominal build
catkin_make -j

# Now overlay w/ virtualenv
python3 -m virtualenv -p python2 --system-site-packages .
./devel/bin/pip install pymycobot

source ./devel/setup.bash
...
rosrun myCobotROS display.py  # Should still work
```

zlj-zz, 2021-05-10:
Can you give me a detailed explanation?It's OK on my computer,no virtual environment is required。
`sudo` and `--user` should not be affected, but they should be installed in different packages sites.
You need to use the pip of python2 to install. By default, ROS kinetic uses python2.

eacousineau, 2021-06-25:
> You can also try to add to your shell [...]

Ermagherd, thank you!!!

> Can you give me a detailed explanation?It's OK on my computer,no virtual environment is required。

I've added a relatively brief explanation here as well:
https://github.com/zlj-zz/minimycobotflasher/issues/1
Just from a quick search, I believe this post echoes my sentiment: https://stackoverflow.com/a/61452956

zlj-zz, 2021-07-05:
I got it, it's better. But the problem is not each user like the vistual env.

This will increase the difficulty of using rookies, and they are also the objects we need to consider.