elephantrobotics/mycobot_ros #21: Adding CI for build testing

Tiryoh, 2021-05-18:
How about adding a CI configuration (e.g. https://github.com/ros-industrial/industrial_ci) to check if catkin_make can be done without errors?
I think you'll notice bugs like #20 right away.
I can send a PR if you wish.

zlj-zz, 2021-05-18:
Of course

Tiryoh, 2021-05-22:
OK! I'll send a PR.

Tiryoh, 2021-05-22:
@zlj-zz Which branch should I send the request to?
What is the difference between `noetic`, `master` and `dev`?

zlj-zz, 2021-05-22:
master

Tiryoh, 2021-05-24:
> why the env is melodic? It's same with kinetic?

ROS Kinetic and Ubuntu 16.04 has been EOL (End of Life).
It's not exactly the same, but it can be used in much the same way, so I'll check if I can add Kinetic.
It could be difficult because it is not supported officially anymore...

> All features are OK？

I haven't tested all, but it seems OK. I don't have a camera for myCobot.

zlj-zz, 2021-05-25:
The PR has been merged and the issue will be closed. Notice here.

closed 2021-05-25 by zlj-zz