elephantrobotics/mycobot_ros #25: can't spawn mycobot model on gazebo

graziegrazie, 2021-05-27:
I tested gazebo.launch on laptop host, not Docker. Mycobot model seems to be spawned, but it is interrupted by someone. My environment is following.
 Ubuntu 20.04
 ROS noetic
Any suggestion is helpful for me.

# console output
$ roslaunch mycobot_ros gazebo.launch 
... logging to /home/yoshi/.ros/log/07e9e666-beed-11eb-8814-61c2beabc12d/roslaunch-carp-101041.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://carp:39985/

SUMMARY

PARAMETERS
 * /controller_list: [{'name': 'mycobo...
 * /gazebo/enable_ros_network: True
 * /generic_hw_control_loop/cycle_time_error_threshold: 0.01
 * /generic_hw_control_loop/loop_hz: 300
 * /hardware_interface/joints: ['joint2_to_joint...
 * /hardware_interface/sim_control_mode: 1
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /moveit_sim_hw_interface/joint_model_group: mycobot_arm
 * /moveit_sim_hw_interface/joint_model_group_pose: init_pose
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /use_sim_time: True

NODES
  /
    controller_spawner (controller_manager/spawner)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_gazebo_model (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [101055]
process[gazebo_gui-2]: started with pid [101060]
process[spawn_gazebo_model-3]: started with pid [101065]
process[controller_spawner-4]: started with pid [101066]
usage: spawner [-h] [--stopped] [--wait-for topic] [--namespace ns] [--timeout T] [--no-timeout]
               [--shutdown-timeout SHUTDOWN_TIMEOUT]
               controller [controller ...]
spawner: error: the following arguments are required: controller
[controller_spawner-4] process has died [pid 101066, exit code 2, cmd /opt/ros/noetic/lib/controller_manager/spawner __name:=controller_spawner __log:=/home/yoshi/.ros/log/07e9e666-beed-11eb-8814-61c2beabc12d/controller_spawner-4.log].
log file: /home/yoshi/.ros/log/07e9e666-beed-11eb-8814-61c2beabc12d/controller_spawner-4*.log
[INFO] [1622123823.927332, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1622123823.930062, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1622123824.082596111]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1622123824.083617142]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1622123824.217451605]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1622123824.218573599]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1622123824.944414549]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1622123824.987983233, 0.039000000]: Physics dynamic reconfigure ready.
[INFO] [1622123825.136705, 0.183000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1622123825.346427, 0.368000]: Spawn status: SpawnModel: Successfully spawned entity
[spawn_gazebo_model-3] process has finished cleanly
log file: /home/yoshi/.ros/log/07e9e666-beed-11eb-8814-61c2beabc12d/spawn_gazebo_model-3*.log

# spawn_gazebo_model-3.log
[rospy.client][INFO] 2021-05-27 22:50:05,347: init_node, name[/spawn_gazebo_model], pid[98985]
[xmlrpc][INFO] 2021-05-27 22:50:05,347: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2021-05-27 22:50:05,348: Started XML-RPC server [http://carp:38075/]
[rospy.impl.masterslave][INFO] 2021-05-27 22:50:05,348: _ready: http://carp:38075/
[rospy.registration][INFO] 2021-05-27 22:50:05,348: Registering with master node http://localhost:11311
[... 92 lines cut ...]
[rospy.internal][INFO] 2021-05-27 22:56:19,070: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2021-05-27 22:56:19,070: topic[/clock] removing connection to http://carp:33613/
[rospy.impl.masterslave][INFO] 2021-05-27 22:56:19,070: atexit
[rospy.client][INFO] 2021-05-27 22:57:03,818: init_node, name[/spawn_gazebo_model], pid[101065]
[xmlrpc][INFO] 2021-05-27 22:57:03,818: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2021-05-27 22:57:03,819: Started XML-RPC server [http://carp:41943/]
[rospy.init][INFO] 2021-05-27 22:57:03,819: ROS Slave URI: [http://carp:41943/]
[rospy.impl.masterslave][INFO] 2021-05-27 22:57:03,819: _ready: http://carp:41943/
[rospy.registration][INFO] 2021-05-27 22:57:03,820: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2021-05-27 22:57:03,820: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2021-05-27 22:57:03,919: registered with master
[rospy.rosout][INFO] 2021-05-27 22:57:03,919: initializing /rosout core topic
[rospy.rosout][INFO] 2021-05-27 22:57:03,923: connected to core topic /rosout
[rospy.simtime][INFO] 2021-05-27 22:57:03,923: initializing /clock core topic
[rospy.simtime][INFO] 2021-05-27 22:57:03,925: connected to core topic /clock
[rosout][INFO] 2021-05-27 22:57:03,927: Loading model XML from ros parameter robot_description
[rosout][INFO] 2021-05-27 22:57:03,930: Waiting for service /gazebo/spawn_urdf_model
[rospy.internal][INFO] 2021-05-27 22:57:04,149: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2021-05-27 22:57:04,952: topic[/clock] adding connection to [http://carp:37935/], count 0
[rosout][INFO] 2021-05-27 22:57:05,136: Calling service /gazebo/spawn_urdf_model
[rosout][INFO] 2021-05-27 22:57:05,346: Spawn status: SpawnModel: Successfully spawned entity
[rospy.core][INFO] 2021-05-27 22:57:05,347: signal_shutdown [atexit]
[rospy.internal][INFO] 2021-05-27 22:57:05,348: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2021-05-27 22:57:05,349: topic[/clock] removing connection to http://carp:37935/
[rospy.impl.masterslave][INFO] 2021-05-27 22:57:05,349: atexit

zlj-zz, 2021-05-30:
The reason why gozabo cannot be started may be due to URDF. The model lacks some parameters required by gozabo. I'm not sure if it will be fixed later, maybe gozabo is not needed.
BTW,This package doesn't seem to run directly on `neotic`.

casiarobot, 2021-07-20:
> The reason why gozabo cannot be started may be due to URDF. The model lacks some parameters required by gozabo. I'm not sure if it will be fixed later, maybe gozabo is not needed.
> BTW,This package doesn't seem to run directly on `neotic`.

确实因为URDF，经过加入collision模块和惯性模块，我的gazebo除了第5部分已经可以显示。
而第五部分的dae文件有问题。

casiarobot, 2021-07-20:
[image]

Tiryoh, 2021-09-04:
I also tried it, but it looks like the collision needs to be changed in the current URDF.
The package, which I made using the URDF released around December 2020, can run Gazebo's myCobot from MoveIt, so I'll share it for your reference.
https://github.com/Tiryoh/mycobot_ros#moveit--gazebo

tongtybj, 2021-12-16:
Based on the work of @Tiryoh  (https://github.com/Tiryoh/mycobot_ros), I have modified lots of files in this repository and finally I restored the gazebo simulation as explained in  https://github.com/elephantrobotics/mycobot_ros/pull/47.

I hope some one help me to test this PR, especially in ros noetic env.

Tiryoh, 2021-12-21:
@tongtybj Great! 
It looks a little dark in Gazebo. I'll add a review comment later.

I have released the repository under the MIT license, please embed the original repository copyright notice and MIT license text according to the MIT license.
For reference, here are some example of my work using other OSS software.

https://github.com/Tiryoh/mycobot_ros/blob/main/README.md?plain=1#L62-L72
https://github.com/Tiryoh/actions-mkdocs/blob/main/README.md?plain=1#L73-L75

tongtybj, 2021-12-28:
@Tiryoh 

Thanks for your contribution!  I have merged your PR.

For license issue, could you also send a PR to #47 if necessary?

carlos-411, 2022-06-03:
> [image]

Just wanted to see how you were able to get two cobots in the same enviorment.
Maybe this is a very simple question, but am having many issues trying to solve that.

casiarobot, 2022-06-07:
> 

You can spawn many times with various namespace or name by 'rosrun gazebo_ros spawn_model '

Help can be got by 'rosrun gazebo_ros spawn_model --help'

closed 2025-04-27 by wangWking