elephantrobotics/mycobot_ros #29: End-effectors urdf/srdf?

TrajM, 2021-07-01:
Hello, I just started using myCobot with ROS and I want to run an application with the suction pump.
I am new to ROS, so I am not sure if the code is already available here. I wanted to ask if there is a file to define the suction pump model and how to attach it to the robot arm?
Thank you

Eheech, 2021-07-02:
Sorry, we don't have any files to define the suction pump model at the moment .

junjungoal, 2021-07-05:
Hi, I would also like you to add an urdf file for an adaptive gripper.
Thank you very much.

zlj-zz, 2021-07-05:
@TrajM @junjungoal The gripper model has been added to the URDF of pro320 model.

Please check [https://github.com/elephantrobotics/mycobot_ros/tree/main/urdf/320_urdf](https://github.com/elephantrobotics/mycobot_ros/tree/main/urdf/320_urdf).😊

junjungoal, 2021-07-05:
@zlj-zz 
that's awesome!! 
Thank you very much for handling it promptly.

closed 2021-11-15 by Eheech