elephantrobotics/mycobot_ros #30: Adaptive Gripper dae broken in noetic

junjungoal, 2021-07-06:
Thank you for adding a dae file for an adaptive gripper recently. It worked in melodic, but it seems to be broken in noetic... could you please check it?  Since i need to use noetic for my project, I would like you to fix it.

junjungoal, 2021-07-06:
@zlj-zz 

According to the output, It couldn't find the dae file though it's correct path (I tried this in Melodic, and this worked).  FYI, I also replace it with the absolute path, but couldn't load it.
The exactly same issue happened to `joint5.dae` before, but it's resolved by replacing the `joint5.dae` in `main` branch with the one in noetic branch (https://github.com/elephantrobotics/mycobot_ros/blob/noetic/urdf/joint5.dae).. I was not sure what caused this at that time,.. but taking this case into consideration, I assume this is not just an issue of path...

```
[ WARN] [1625557548.270360601]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://mycobot_ros/urdf/320_urdf/gripper_ag.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 753)
[ERROR] [1625557548.270412671]: Could not load model 'package://mycobot_ros/urdf/320_urdf/gripper_ag.dae' for link 'gripper': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://mycobot_ros/urdf/320_urdf/gripper_ag.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 753)
```

Also, it seems that `joint5.dae` was fixed for noetic before ... and this might be relevant 
(https://github.com/elephantrobotics/mycobot_ros/commit/59cbfcbd8decad3004e0e2e28b18d2579eec2a64)

junjungoal, 2021-07-07:
I could not load all of the dae of mycobot_320 in noetic as well, but it's successful in melodic.. and I got the same errors as https://github.com/elephantrobotics/mycobot_ros/issues/30#issuecomment-874547372.

zlj-zz, 2021-07-07:
I see you seem to use gripper for mycobot, but this model is about pro320.

junjungoal, 2021-07-08:
@zlj-zz  oh, pro320 did not work as well in noetic env due to the same error..

Interestingly, this issue is  resolved by exporting the 3d model as a dae file from blender... you could check https://github.com/junjungoal/mycobot_ros/blob/main/urdf/gripper_ag.dae

I assume you export the model from rhino 3d..? and it seems to be broken in noetic env...
I think it's worth investigating this more later time.

zlj-zz, 2021-07-08:
How to solve it? Import the model into blender and re export it?

junjungoal, 2021-07-08:
I firstly converted the gripper_ag.dae to gripper_ag.obj since I could not import the dae file properly in Blender. 
Then, after the conversion, I imported the obj file in Blender and export it as a new dae file.

zlj-zz, 2021-07-08:
how to converte？ blender？

I will tell my colleagues about this model.

junjungoal, 2021-07-08:
I used this free online converter https://3d-convert.com/en/convert/dae-to-obj.html
but, we could use any converter as long as it works..
Thanks!

zlj-zz, 2021-07-08:
Thank you very much for your support.🙏

closed 2021-07-08 by junjungoal