elephantrobotics/mycobot_ros #36: Connecting to real robot

saratrajput, 2021-09-07:
Hello. This might sound like a naive question but I'm not able to figure this out from the documentation.

I have a myCobot-pi model and it has ROS-Kinetic pre-installed on it.

When I run:
`rosrun mycobot_ros slider_control.py`

I get the error: unable to connect to "/dev/ttyUSB0"

My question is if I'm running ROS on the robot's RPi, do I need to specify the port?

Secondly, if I'm running ROS core on my laptop instead, don't I just need to specify the IP address of the ROS core machine for the robot's nodes on the RPi? Why do I need to connect with a USB cable?

Please let me know if I'm missing something or point me to some good documentation.

Eheech, 2021-09-18:
Have you tried ' /dev/ttyAMA0 ' ?

Eheech, 2021-09-18:
> My question is if I'm running ROS on the robot's RPi, do I need to specify the port?
> Re : Have to specify the port if you want to connect the machine
> Secondly, if I'm running ROS core on my laptop instead, don't I just need to specify the IP address of the ROS core machine for the robot's nodes on the RPi? Why do I need to connect with a USB cable?
> 
> Please let me know if I'm missing something or point me to some good documentation.

Eheech, 2021-09-29:
Also, you can tried ' sudo chmod 777 /dev/ttyUSB* ' , hope this commang would help you.

blcuwc, 2022-02-23:
> Hello. This might sound like a naive question but I'm not able to figure this out from the documentation.
>
> I have a myCobot-pi model and it has ROS-Kinetic pre-installed on it.
>
> When I run: `rosrun [...]

Please make sure that you follow the tutorial correcttly : https://www.elephantrobotics.com/docs/myCobot-en/3-development/4-ros&moveit/

These two commands should be executed in independent terminal windows before you run the `rosrun` command:

`roscore` and `roslaunch mycobot_ros mycobot_slider.launch`

Then run the `rosrun mycobot_ros slider_control.py` command in another terminal window.

Also you should check the connection between your laptop and the mycobot RPi, make sure the ports and the cable are in good condition.

As for the IP address, because of the serial interface is used in the connection between the laptop and robot, specifying the IP address is not necessary to the recognition and doesn't make sense.

Hope your problem has been addressed.

closed 2022-02-24 by blcuwc