elephantrobotics/mycobot_ros #37: Error when building docker container

saratrajput, 2021-09-11:
I have the following error when building the docker container.

```
Building ros
Sending build context to Docker daemon  190.1MB
Step 1/16 : ARG BASE_IMAGE
Step 2/16 : FROM ${BASE_IMAGE}
18.04: Pulling from library/ubuntu
e4ca327ec0e7: Pull complete 
Digest: sha256:9bc830af2bef73276515a29aa896eedfa7bdf4bdbc5c1063b4c457a4bbb8cd79
Status: Downloaded newer image for ubuntu:18.04
 ---> 54919e10a95d
Step 3/16 : COPY --from=osrf/ros:melodic-desktop-full / /
melodic-desktop-full: Pulling from osrf/ros
4bbfd2c87b75: Pull complete 
d2e110be24e1: Pull complete 
889a7173dcfe: Pull complete 
1270d8e5a641: Pull complete 
6b8ca49cb51a: Pull complete 
c95c6800a35c: Pull complete 
47408d20bf58: Pull complete 
29f56ec2afd1: Pull complete 
a07d96849f4e: Pull complete 
86b45c5e6466: Pull complete 
d761800cb971: Pull complete 
8f3409eabec0: Pull complete 
aebe9f16eba2: Pull complete 
5c80026661cc: Pull complete 
d629bf3d1a44: Pull complete 
Digest: sha256:f9268872571969d1fa20a64d1a9225dd592451862b21c34d66e7b6be3541b8dd
Status: Downloaded newer image for osrf/ros:melodic-desktop-full
 ---> 457eabb12615
Step 4/16 : ENV BASH_ENV="/root/launch.sh"
 ---> Running in a2fba33a5c2a
Removing intermediate container a2fba33a5c2a
 ---> 400c387c6f8f
Step 5/16 : SHELL ["/bin/bash", "-c"]
 ---> Running in d478ec733dfb
Removing intermediate container d478ec733dfb
 ---> 1b56a0c4ae8d
Step 6/16 : ENTRYPOINT ["/bin/bash", "-c"]
 ---> Running in a599a61cbb3d
Removing intermediate container a599a61cbb3d
 ---> d4a0351a3d17
Step 7/16 : ARG ROS_DISTRO
 ---> Running in 57bc2b5de5cb
Removing intermediate container 57bc2b5de5cb
 ---> 6e04c0dc7082
Step 8/16 : RUN echo "source /opt/ros/${ROS_DISTRO}/setup.bash" >> $BASH_ENV
 ---> Running in 1adec7b2084b
Removing intermediate container 1adec7b2084b
 ---> 5abb05b78021
Step 9/16 : WORKDIR /catkin_ws/src
 ---> Running in 67decfcc552b
Removing intermediate container 67decfcc552b
 ---> 4f4de8e20a54
Step 10/16 : COPY . mycobot_ros
 ---> 1d56f290fbfb
Step 11/16 : RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add - &&     apt-get update &&     apt-get install -y         python-rosinstall         python-rosinstall-generator         python-wstool         build-essential         python-pip &&     rosdep install -r -y -i --from-paths . &&     rm -rf /var/lib/apt/lists/*
 ---> Running in ea7ef011be0e
Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
Get:1 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:2 http://packages.ros.org/ros/ubuntu bionic InRelease [4680 B]
Get:3 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
Get:4 http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages [772 kB]
Get:5 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [26.7 kB]
Get:6 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [2320 kB]
Get:7 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:8 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
[... 526 lines cut ...]
 ---> ebc79e332092
Step 12/16 : ARG PYMYCOBOT_VERSION
 ---> Running in cea6ee5459b7
Removing intermediate container cea6ee5459b7
 ---> d29a06c5bb6b
Step 13/16 : RUN pip install "pymycobot $PYMYCOBOT_VERSION" --user
 ---> Running in d38a7a7eb23a
Collecting pymycobot>=2<3
  Downloading https://files.pythonhosted.org/packages/f6/2e/766270044b6ec965e4fc37e908578497b402452cd94f755e87c8686e1263/pymycobot-2.6.0b1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-6DbU8C/pymycobot/setup.py", line 12, in <module>
        import pymycobot
      File "pymycobot/__init__.py", line 9, in <module>
        from pymycobot import utils
      File "pymycobot/utils.py", line 7, in <module>
        except ModuleNotFoundError:
    NameError: name 'ModuleNotFoundError' is not defined
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6DbU8C/pymycobot/
The command '/bin/bash -c pip install "pymycobot $PYMYCOBOT_VERSION" --user' returned a non-zero code: 1
ERROR: Service 'ros' failed to build : Build failed
```

abrahamperz, 2021-09-17:
Same issue, at same **13**th **step**

apockill, 2021-09-20:
Maybe we should pin the `pymycobot` version in the docker container? :thinking:

abrahamperz, 2021-09-20:
@apockill could be, I am using Ubuntu 18.04 LTS
Also if you want to share the dockerfile I can help you with the debugging

apockill, 2021-09-20:
I've created a PR that should fix the issue https://github.com/elephantrobotics/mycobot_ros/pull/39/files

I think we might want to add some CI to the repo that tests the builds complete.

abrahamperz, 2021-09-20:
@apockill the error stills  in the docker image
Step 1/16 : ARG BASE_IMAGE
Step 2/16 : FROM ${BASE_IMAGE}
 ---> 4f4bf5e1fe4b
Step 3/16 : COPY --from=osrf/ros:melodic-desktop-full / /
 ---> Using cache
 ---> cb685401d019
Step 4/16 : ENV BASH_ENV="/root/launch.sh"
 ---> Using cache
 ---> b2bbecebf2db
Step 5/16 : SHELL ["/bin/bash", "-c"]
 ---> Using cache
 ---> bd921b40ad6f
Step 6/16 : ENTRYPOINT ["/bin/bash", "-c"]
 ---> Using cache
 ---> 6610e64dae57
Step 7/16 : ARG ROS_DISTRO
 ---> Using cache
 ---> 0c29316b2a5e
Step 8/16 : RUN echo "source /opt/ros/${ROS_DISTRO}/setup.bash" >> $BASH_ENV
 ---> Using cache
 ---> 799171fcaf54
Step 9/16 : WORKDIR /catkin_ws/src
 ---> Using cache
 ---> f7842b054bdc
Step 10/16 : COPY . mycobot_ros
 ---> c04de159f91c
Step 11/16 : RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add - &&     apt-get update &&     apt-get install -y         python-rosinstall         python-rosinstall-generator         python-wstool         build-essential         python-pip3 &&     rosdep install -r -y -i --from-paths . &&     rm -rf /var/lib/apt/lists/*
 ---> Running in 35cf8a2aa84c
[91mWarning: apt-key output should not be parsed (stdout is not a terminal)
[0mOK
Get:1 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:2 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
Get:3 http://packages.ros.org/ros/ubuntu bionic InRelease [4680 B]
Get:4 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [1428 kB]
Get:5 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [26.7 kB]
Get:6 http://security.ubuntu.com/ubuntu bionic-security/multiverse i386 Packages [7114 B]
Get:7 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [567 kB]
Get:8 http://security.ubuntu.com/ubuntu bionic-security/restricted i386 Packages [26.7 kB]
Get:9 http://security.ubuntu.com/ubuntu bionic-security/main i386 Packages [1296 kB]
Get:10 http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages [772 kB]
Get:11 http://security.ubuntu.com/ubuntu bionic-security/universe i386 Packages [1238 kB]
Get:12 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [2326 kB]
Get:13 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:14 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:15 http://archive.ubuntu.com/ubuntu bionic/multiverse i386 Packages [177 kB]
Get:16 http://archive.ubuntu.com/ubuntu bionic/restricted amd64 Packages [13.5 kB]
Get:17 http://archive.ubuntu.com/ubuntu bionic/main amd64 Packages [1344 kB]
Get:18 http://packages.ros.org/ros/ubuntu bionic/main i386 Packages [26.3 kB]
Get:19 http://archive.ubuntu.com/ubuntu bionic/restricted i386 Packages [13.5 kB]
Get:20 http://archive.ubuntu.com/ubuntu bionic/multiverse amd64 Packages [186 kB]
Get:21 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [11.3 MB]
Get:22 http://archive.ubuntu.com/ubuntu bionic/main i386 Packages [1328 kB]
Get:23 http://archive.ubuntu.com/ubuntu bionic/universe i386 Packages [11.3 MB]
Get:24 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [2202 kB]
Get:25 http://archive.ubuntu.com/ubuntu bionic-updates/main i386 Packages [1688 kB]
Get:26 http://archive.ubuntu.com/ubuntu bionic-updates/restricted i386 Packages [37.2 kB]
Get:27 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [600 kB]
Get:28 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse i386 Packages [13.1 kB]
Get:29 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [34.4 kB]
Get:30 http://archive.ubuntu.com/ubuntu bionic-updates/universe i386 Packages [1985 kB]
Get:31 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [2761 kB]
Get:32 http://archive.ubuntu.com/ubuntu bionic-backports/main amd64 Packages [11.3 kB]
Get:33 http://archive.ubuntu.com/ubuntu bionic-backports/universe i386 Packages [11.4 kB]
Get:34 http://archive.ubuntu.com/ubuntu bionic-backports/main i386 Packages [11.2 kB]
Get:35 http://archive.ubuntu.com/ubuntu bionic-backports/universe amd64 Packages [11.4 kB]
Fetched 43.3 MB in 4s (10.4 MB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
[91mE: Unable to locate package python-pip3
[0m

apockill, 2021-09-21:
How are you running the build?

apockill, 2021-09-21:
[image]
@abrahamperz it appears you didn't check out my PR, and typed it yourself. The package name is `python3-pip` not `python-pip3`

closed 2021-09-24 by fedorenchik