elephantrobotics/mycobot_ros #40: ROS in docker: rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException)

nakano16180, 2021-09-24:
Error:

```
 [ WARN] [1632475519.441226059]: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61)
ros_1         | rviz::RenderSystem: error creating render window: OGRE EXCEPTION(3:RenderingAPIException): Unable to create a suitable GLXContext in GLXContext::GLXContext at /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/RenderSystems/GL/src/GLX/OgreGLXContext.cpp (line 61)
```

Environment:

```
$ docker-compose --version --short
docker-compose version 1.29.2, build 5becea4c

$ docker --version
Docker version 20.10.8, build 3967b7d

$ dpkg -l | grep nvidia-docker2
ii  nvidia-docker2                                  2.6.0-1 

$ nvidia-smi | head -n 4
Fri Sep 24 18:27:49 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.91.03    Driver Version: 460.91.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+

```

I see [this issue](https://github.com/NVIDIA/nvidia-docker/issues/1438#issuecomment-846535031).
Adding the followings to the ```docker-compose.yaml``` solved the problem, but I can't run docker container without GPU.

```diff
x-app: &common
  command: [ "roslaunch mycobot_320 mycobot_320_slider.launch" ]
  privileged: true
  environment:
    PYTHONUNBUFFERED: 1
    QT_X11_NO_MITSHM: 1
    DISPLAY: $DISPLAY
+    NVIDIA_VISIBLE_DEVICES: all
+    NVIDIA_DRIVER_CAPABILITIES: all
  volumes:
    - "/tmp/.X11-unix:/tmp/.X11-unix:rw"
  build:
    context: .
    args: &common-args
      ROS_DISTRO: "melodic"
      PYMYCOBOT_VERSION: ">=2<3"
      BASE_IMAGE: nvidia/opengl:1.1-glvnd-runtime-ubuntu18.04

```

apockill, 2021-09-24:
What command are you using to start the container?

nakano16180, 2021-09-24:
I use following command

```
docker-compose build ros && xhost +local:root && docker-compose up ros
```

apockill, 2021-09-24:
Gotcha. Can you try the other command (it's from the readme)
```bash
docker-compose build nvidia-ros && xhost +local:root && docker-compose up nvidia-ros
```

nakano16180, 2021-09-24:
I tried that command, it's working.

closed 2022-02-24 by blcuwc