elephantrobotics/mycobot_ros #43: mycobotpi catkin_make error

yo4hi6o, 2021-11-04:
Hi
If you change the ros_catkin_ws originally in mycobotpi and run the catkin_make, the following error occurs. Is there a way to solve this?
```
pi@raspberrypi:~/ros_catkin_ws $ catkin_make
Traceback (most recent call last):
  File "/home/pi/ros_catkin_ws/devel/bin/catkin_make", line 7, in <module>
    with open(python_script, 'r') as fh:
IOError: [Errno 2] No such file or directory: '/home/pi/ros_catkin_ws/src/catkin/bin/catkin_make'

```

Eheech, 2021-11-05:
What changes did you make to os_catkin_ws ? Maybe you can upload the review process ,  let us check .

Eheech, 2021-11-05:
If you follow the tutorial in our gitbook , will it return this error ?

yo4hi6o, 2021-11-05:
I just changed rosrun mycobot_ros slider_control.py from the issue(#36 ) to / dev / ttyAMA0.
Also, the same error occurred in catkin_make when creating a new catkin_ws described in the README of micobotpi.
Currently, I don't have this machine at hand, so I can contact you for details on Monday.

Eheech, 2021-11-05:
Have you tried ' catkin_init_workspace ' , bafore you used ' catkin_make ' ?

yo4hi6o, 2021-11-05:
> Have you tried ' catkin_init_workspace ' , bafore you used ' catkin_make ' ?

OK,
I will try it once.

Eheech, 2021-11-05:
In this tutorial : https://www.elephantrobotics.com/docs/myCobot-en/3-development/4-ros&moveit/2-install_ros_package.html
before  ' **Installation** '  you can use these commands :
[image]
and then :
[image]

yo4hi6o, 2021-11-05:
OK,
I don't have the actual machine right now, so I will try it at the beginning of the week and report it.
Thank you

Eheech, 2021-11-05:
You're welcom, hope i can help you .

yo4hi6o, 2021-11-08:
Hi
A similar error occurs, and after purchase, no changes have been made at the system location. I've done a lot of WS creation in regular ROS and MyCobot, but do I need any special initial setup for MyCobotpi?

```
pi@raspberrypi:~ $ mkdir -p ~/catkin_ws/src
pi@raspberrypi:~ $ cd ~/catkin_ws/src
pi@raspberrypi:~/catkin_ws/src $ catkin_init_workspace
Traceback (most recent call last):
  File "/home/pi/ros_catkin_ws/devel/bin/catkin_init_workspace", line 7, in <module>
    with open(python_script, 'r') as fh:
IOError: [Errno 2] No such file or directory: '/home/pi/ros_catkin_ws/src/catkin/bin/catkin_init_workspace'
pi@raspberrypi:~/catkin_ws/src $ 
```

Eheech, 2021-11-08:
Have you tired to use ' catkin_init_workspace ' in derectory ' ros_catkin_ws ' ?

yo4hi6o, 2021-11-08:
Hi,
It is m same in ros_catkin_ws.
thank you
```
pi@raspberrypi:~ $ cd ros_catkin_ws
pi@raspberrypi:~/ros_catkin_ws $ catkin_init_workspace
Traceback (most recent call last):
  File "/home/pi/ros_catkin_ws/devel/bin/catkin_init_workspace", line 7, in <module>
    with open(python_script, 'r') as fh:
IOError: [Errno 2] No such file or directory: '/home/pi/ros_catkin_ws/src/catkin/bin/catkin_init_workspace'
pi@raspberrypi:~/ros_catkin_ws $ 
```

Eheech, 2021-11-08:
Which Ros do you install in Raspberry ? Noetic or Kinetic ? If you are not sure , you can use command ' roscore ' to check .

yo4hi6o, 2021-11-08:
It is kinetic that was originally installed.

Eheech, 2021-11-08:
Would you mind offer a screenshot of your system ?

Eheech, 2021-11-08:
Please check ~/.bashrc and ~/.zshrc , make sure you have added ros path .

yo4hi6o, 2021-11-08:
[image]

Eheech, 2021-11-08:
Please use these two commands ' cat ~/.bashrc | grep 'ros' ' and ' cat ~/.zshrc '  and send screenshot to us , we wanna find out what the problem it is , thank you very much !!!!

yo4hi6o, 2021-11-08:
PATH is in .bash.
Why do I get such an error
I'm curious.

[image]

Eheech, 2021-11-08:
Please try this way , thank you
[image]

yo4hi6o, 2021-11-09:
OK, thank you. For a business trip until the day after tomorrow I will try it the day after tomorrow. thank you!

yo4hi6o, 2021-11-12:
I was able to resolve the error by changing this path.
Thank you very much.

Eheech, 2021-11-15:
You're welcome, we're delighted to help you.

closed 2021-11-15 by Eheech