elephantrobotics/mycobot_ros #45: How to release the end effector from leveling.

yo4hi6o, 2021-11-23:
Hi.
Is this a hardware issue with MyCobot?
We are currently using a 6-axis robot arm (MyCobot) controlled by MoveIt and with the addition of a gripper. Moveit's automatic route generation cannot sample the valid state of the goal tree if error: RRTConnect: z = 0.15 or less. I want to grab something close to the ground. As you can see in this image, I think one of the factors is that the angle of the 4th axis does not change downward. After investigating, I think that the function to keep the position of the end effector horizontal is working. Is there a way to unlock it and use Move to bring the end effector closer to the ground? We have confirmed that the designated position is a position where the target can be achieved.

[image]

pick_and_placer.cpp

```cpp
#include <ros/ros.h>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char **argv) {
  ros::init(argc, argv, "pickandplacer");
  ros::NodeHandle nh;

  ros::AsyncSpinner spinner(2);
  spinner.start();

  moveit::planning_interface::MoveGroupInterface arm("mycobot_arm");

  arm.setPoseReferenceFrame("base_link");

  ROS_INFO("Moving to prepare pose");
  geometry_msgs::PoseStamped pose1;
  pose1.header.frame_id = "base_link";
  pose1.pose.position.x = 0.0;
  pose1.pose.position.y = 0.25;
  pose1.pose.position.z = 0.15;

  moveit::planning_interface::MoveItErrorCode ret;

  ROS_INFO("move to WP1");
  arm.setPoseTarget(pose1);
  ret = arm.move();
  if (!ret) {
    ROS_WARN("Fail: %i", ret.val);
    arm.setNamedTarget("vertical");
    arm.move();
  }
  ros::Duration(1.5).sleep();

  ros::shutdown();
  return 0;
}

```

URDF

```xml
<?xml version="1.0" ?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="firefighter"
       xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller"
       xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface"
       xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor">

    <xacro:property name="width" value=".2" />
    <material name="dark blue">
        <color rgba="0.1 0.1 0.9 1" />
    </material>
    <material name="gray">
        <color rgba="0.8 0.8 0.8 1" />
    </material>
    <link name="world"/>

  <link name="base_link"/>
  <joint name="world_to_base_link" type="fixed">
    <parent link="world"/>
    <child link="base_link"/>
[... 319 lines cut ...]
    </joint>

<!-- camera holder joint & link -->

<xacro:include filename="$(find gazebo_sensors_tutorial)/robot/sensors/cam_holder.urdf.xacro"/>

    <joint name="camera_holder_joint" type="fixed">
    <parent link="base_link" />
    <child link="camera_holder_link" />
    <origin xyz="0 -0.50 0" rpy="1.5705 0 0"/>
    </joint>

<link name="camera_holder_link" />

  <xacro:cam_holder name="camera_holder" parent="base_link">
    <origin xyz="0 -0.17 1.0" rpy="0 0 1.5705"/>
  </xacro:cam_holder>

    <gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
            <robotNamespace>/</robotNamespace>
        </plugin>
    </gazebo>

```

anla-xu, 2021-11-24:
@yo4hi6o It looks like the end effector is very close to the ground in the picture, doesn't it? Can you provide a more detailed description? I cannot understand your needs from your expression.

blcuwc, 2022-02-23:
@yo4hi6o It is good that you confirmed the designed position is avaliable for the robot, and this shows that it is less possible to be a hardware issue of the robot.

According to the error log you provided ` RRTConnect: z = 0.15 or less` and your code `pose1.pose.position.z = 0.15;`, it seems that there is something wrong with position z setting. Please try other values of position z setting, and see if it's possible to reach your goal. Otherwise, check the corresponding URDF file and exam the whole moving process.

Hope this helps.

closed 2022-02-24 by blcuwc