elephantrobotics/mycobot_ros #113: Can i request for an update of the Mycobot Jetson Nano URDF directory to include the xacro and stl files?

abaddonn, 2023-02-02:
Could I request for the STL files for each joint of the mycobot Jetson Nano version to be included in the package and the urdf.xacro to be updated to include collisions? Much appreciated.

Heres some context, my environment is Ubuntu 20.04 and ROS noetic. I am trying to utilise the mycobot Jetson Nano for dynamic obstacle avoidance. I am utilising the mycobot Jetson Nano version with moveit. however I noticed that under the mycobot_description, there is no mesh file or .stl files for collision. While using the moveit setup assistant and loading up the mycobot_urdf.urdf to create a new moveit config i am unable to add collisions as well.

Executing the demo.launch prompts me the following error below specifying to: Fix your URDF file by explicitly specifying collision geometry"

I tested gazebo.launch and this error below is prompted. Additionally, the robot is not spawned in the gazebo environment.

**Demo.launch**
`
[ WARN] [1675239446.607871979]: Link joint6 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.607646616]: Link joint1 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.607757578]: Link joint2 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.607778201]: Link joint3 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.607827984]: Link joint4 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.607860146]: Link joint5 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.607881020]: Link joint6_flange has visual geometry but no collision geometry. Collision geometry will be left empty. `

`
Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1675239446.608061800]: No geometry is associated to any robot links
`
[image]

**Gazebo.launch**

`[INFO] [1675236096.385363, 0.000000]: Calling service /gazebo/spawn_urdf_model
Error [parser.cc:488] parse as old deprecated model file failed.
Error Code 4 Msg: Required attribute[filename] in element[plugin] is not specified in SDF.
Error Code 8 Msg: Error reading element <plugin>
Error Code 8 Msg: Error reading element <model>
Error Code 8 Msg: Error reading element <sdf>
`

[image]

wangWking, 2024-03-06:
1. 目前的urdf使用的是.dae格式的网格文件，不是.stl文件
2. 目前的Ubuntu 20.04 ROS1 NOETIC系统在Gazebo中已经不支持.urdf格式的模型文件，改为支持.sdf格式，你这边需要重新将urdf改写为sdf格式

wangWking, 2025-04-27:
In Ubuntu 20.04, the model used by gazebo may not support the urdf format, but supports the sdf format instead. Therefore, gazebo recommends using Ubuntu 18.04.

closed 2025-04-27 by wangWking